pimox/pimox7 #122: Create CT: No content

eyduh, 2023-02-08:
When creating a container from the Web UI I get this after pressing create in the final step:

[image]

Running pimox on an Odroid HC4

jiangcuo, 2023-02-19:
can you create ct using `pct` command
https://github.com/jiangcuo/Proxmox-Arm64/wiki/Linux-Container

eyduh, 2023-02-27:
hmm. it stalled at some point in the process, in the web gui I can see the container appear with the lock like it usually does prior to being fully setup. when I killed the process it disappeared again.

closed 2023-02-28 by eyduh

eyduh, 2023-02-28:
After a slightly messy (due to Odroid's terrible petitboot) rebase to your repo I am able to create LXCs both from the terminal with `pct` and through the web interface. The webinterface still complains about ssh key verification soI just copy my key after container creation instead of in the web gui