pimox/pimox7 #123: Trying to get Pimox7 working Orange Pi 5

terrorpup, 2023-02-09:
Hi, I know the project states that it's for the Raspberry Pi 4. I am trying to get it working for the Orange Pi 5 since it has a two boards that would great for the this project. 16 GB and 32 GB, with the Oct Core would make this an awesome project. 

The issue I am running into, ceph-dkms, it's giving an error 10, because of that proxmox-ve isn't delpoying. 

I can look into what the error 10, but I figure I would ask. 

I did install kernel-headers 

apt install linux-libc-dev-arm64-cross

Here is what I am using 
Armbian 22.11.4 Bullseye with Linux 5.10.110-rockchip-rk3588

PRETTY_NAME="Armbian 22.11.4 Bullseye"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

I am running this on a NVMe, Samsung 2230 256 GB

Disk /dev/nvme0n1: 238.47 GiB, 256060514304 bytes, 500118192 sectors
Disk model: PM991a NVMe Samsung 256GB
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 16384 bytes / 131072 bytes
Disklabel type: gpt
Disk identifier: C655DFAA-D54B-154E-A49C-CA706BA90CDC

Device          Start       End   Sectors   Size Type
/dev/nvme0n1p1  32768    557055    524288   256M Linux extended boot
/dev/nvme0n1p2 557056 495091712 494534657 235.8G Linux filesystem

root@traveller:~# uname -a
Linux traveller.magidesign.com 5.10.110-rockchip-rk3588 #trunk.0184 SMP Fri Jan 20 18:24:04 UTC 2023 aarch64 GNU/Linux
 
crw-rw---- 1 root kvm      10, 232 Feb  9 10:08 kvm
 
 Before I tried to use pimox, I was able to get libvirtd working with virt-management.  
 
This is error I get, and again I know that this for raspberry pi  4, but this would great for the Orange Pi 5. 

 apt --fix-missing install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up ceph-dkms (0.0.2) ...
Removing old ceph-dkms-0.0.2 DKMS files...

------------------------------
Deleting module version: 0.0.2
completely from the DKMS tree.
------------------------------
Done.
Loading new ceph-dkms-0.0.2 DKMS files...
Building for 5.10.110-rockchip-rk3588
Building initial module for 5.10.110-rockchip-rk3588
Error! Bad return status for module build on kernel: 5.10.110-rockchip-rk3588 (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
dpkg: error processing package ceph-dkms (--configure):
 installed ceph-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of proxmox-ve:
 proxmox-ve depends on ceph-dkms; however:
  Package ceph-dkms is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 ceph-dkms
 proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)

terrorpup, 2023-02-09:
FYI, I did rebuild and downloaded the correct headers, 

linux-headers-legacy-rockchip-rk3588_1.0.8_arm64.deb

Maybe if there was a flag now to use ceph, that might help.  Part of the problem I also saw what it trying to replace the kernel and can't.  It still gives that error exit status 10

root@traveller:~# apt --fix-missing install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up ceph-dkms (0.0.2) ...
Removing old ceph-dkms-0.0.2 DKMS files...

------------------------------
Deleting module version: 0.0.2
completely from the DKMS tree.
------------------------------
Done.
Loading new ceph-dkms-0.0.2 DKMS files...
Building for 5.10.110-rockchip-rk3588
Building initial module for 5.10.110-rockchip-rk3588
Error! Bad return status for module build on kernel: 5.10.110-rockchip-rk3588 (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
dpkg: error processing package ceph-dkms (--configure):
 installed ceph-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of proxmox-ve:
 proxmox-ve depends on ceph-dkms; however:
  Package ceph-dkms is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 ceph-dkms
 proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)

jiangcuo, 2023-02-19:
```
rm -rf /var/lib/dkms/ceph-dkms/
dpkg --configure -a`
```
You can also try my project https://github.com/jiangcuo/Proxmox-Arm64

ShireCiel, 2023-02-21:
Linux Version 5.10.100
Looks i fixed this error.
First step:
Edit: /var/lib/dkms/ceph-dkms/0.0.2/source/default-src/fs/ceph/xattr.c
In: function: ceph_get_xattr_handler
Add: "int flags" to function args

Second step:
Edit: /var/lib/dkms/ceph-dkms/0.0.2/source/default-src/fs/ceph/cache.h
Add: "MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);" in you head of this file.

But still other errors in configure proxmox-ve

ShireCiel, 2023-02-21:
Finished install.
if your hostname is pve make sure you have "pve.proxmox.com" in your hosts.
pve is your hostname. [hostname].proxmox.com