pimox/pimox7 #126: enable wifi of pi4

mirko8054, 2023-03-02:
Hey everyone, on my pi4 i have pimox installed along with its VMs, by default the wifi is disabled with rfkill, which i manually re enable. The goal is to use my raspberry (besides proxmox ve) as router AP, gets internet through ethernet and comes out from wifi as access point. following a guide, once i setupped all the settings, i am unable to set link up for the wlan0, literally doesnt work, any solution?

avlisiur78, 2023-05-07:
Hi, use a external Wi-Fi card, and passthrough the usb to the VM, for now I believe it's not possible to share the internal Wi-Fi card.

varoudis, 2023-06-21:
on-board WIFI works. I run it in AP mode on 5Ghz/80Mhz and I have bridged it with ETH. I run an openwrt that offers DHCP and a inet gateway to the bridge via a USB WIFI card (USB passthrough).

automationxpert, 2023-07-26:
> on-board WIFI works. I run it in AP mode on 5Ghz/80Mhz and I have bridged it with ETH. I run an openwrt that offers DHCP and a inet gateway to the bridge via a USB WIFI card (USB passthrough).

Could you share us the steps to use the on-board Wifi in raspberry Pi.

VocanicZ, 2024-10-11:
I randomly found the way. Please follow these steps.

Check your pci network hardware using `lshw`   
      
    lshw -C network
output are something like this

    *-network:0        
        description: Ethernet interface
        physical id: 1
        logical name: eth0
        serial: xx:xx:xx:xx:xx:xx
        size: 1Gbit/s
        capacity: 1Gbit/s
        capabilities: ethernet physical tp
        configuration: xxxxxxxxxxxxxxxx
    *-network:1 DISABLED        
        description: Wireless interface
        physical id: 2
        logical name: wlan0
        serial: xx:xx:xx:xx:xx:xx
        capabilities: ethernet physical wireless
        configuration: xxxxxxxxxxxxxxxx
As you can see `Wireless interface` are marked as `DISABLED`

Now mark the interface as active by using `ifconfig`

    ifconfig <logical name> up
in my case `logical name` was `wlan0`

execute `lshw -C network` again to check that `DISABLED` are gone

If your wireless device are blocked, enable it  using `rfkill` and try `ifconfig <logical name> up` again

    rfkill unblock wifi

If everything set up correctly, you can be able to connect to the wifi even if ethernet was on.

    nmcli dev wifi list

if you got `Error: NetworkManager is not running.` execute this and try `nmcli dev wifi list` again

    systemctl start NetworkManager
output are something like this

    IN-USE  BSSID              SSID                   MODE   CHAN  RATE        SIGNAL  BARS  SECURITY  
            xx:xx:xx:xx:xx:xx  NathanielSong          Infra  44    270 Mbit/s  100     ▂▄▆█  WPA1 WPA2 
            xx:xx:xx:x:xxx:xx  VocanicZ_5G            Infra  52    270 Mbit/s  97      ▂▄▆█  WPA1 WPA2 
Now you can just easily connect to the SSID using `nmcli`

    nmcli dev wifi connect <SSID> password "<Pass>"

All done. If you want to bridge this WLAN over vm/lxc just configure `NAT` by following [this](https://pve.proxmox.com/wiki/WLAN)