pimox/pimox7 #135: I can't login after install

Woolfy025, 2023-05-08:
Hi. I was trying to get pimox installed on my pi 4 (8GB) and cant get past the login screen. I'm following this video:

https://www.youtube.com/watch?v=g1FXZIvVkq4

When the pi restarts and I try to login with the IP address on the screen, it never works. Not just that but I lose the ability to ssh as well. IP address is still the same previously to installing this. This is the version of OS I have installed: raspberry pi os live 64-bit

I've read about a fix for it by installing debian instead of the other OS. Anyone tried it? I cant seem to find the correct debian version for it. I really don't know what else to try. I've reinstalled it 5 times so far, thinking I didn't do something correctly. As far as I can see, I did everything what the video mentions.

Just seeing if anyone had issues like this and, were able to fix it.

Woolfy025, 2023-05-10:
I was able to get it sorted out. Just an error on my part.

closed 2023-05-10 by Woolfy025

366Cobra, 2023-05-27:
i have the same issue.

i have done a test install, but lost all connection to the pi.
after OS install, ssh was fine, but now, i can no longer use ssh (kex_exchange_identification: read: Connection reset by peer) and going to the site, i just get "An error occurred during a connection to 192.168.2.218:8006. PR_CONNECT_RESET_ERROR"

when installing pimox, i used the same IP as it already had.  there were a couple of errors, but watching install vids, they had the same ones and it worked for them.
i can login and then use the site locally, but no longer over the network.
proxmox does not show entries in the firewall settings and ssh status is up.

i even tried setting the IP for ssh to the one supplied at the install.
port 22 is open and ssh is listening on it.

also set iptable rules for full access.
logs are not showing anything when trying to access the box, so guess traffic is not even let in.
i am a bit rusty what linux is concerned so have not been able to work out what the issue