pimox/pimox7 #143: Install progress gets stuck on 84%

GitMC-8383, 2023-06-09:
I am trying to install PIMox to make my pi useful but the install bar gets stuck on 84%. Will anyone give me a answer to this phemomenon?

GitMC-8383, 2023-06-09:
I followed all of the steps of the guide for the raspberry pi os option until the install script run. It goes through and downloads everything and doing its stuff until 84%. I am on a Raspberry Pi 3 Model B+. I changed the line "if [ $RPIMOD == 3 ]" on line 35 to,  "if [ $RPIMOD == 4 ]" inside of nano then rebooted just to be extra sure it saved, checked it again and the save was there. Then I ran the installer script using "./RPiOS64-IA-Install.sh", it goes through and filled out the prompts and it installed but then it froze at 84%. I tried ssh and monitor, mouse, and keyboard setups but none of them worked.

kevin-brown, 2023-06-10:
I also ran into it stopping at 84% but after 10ish minutes of waiting and making sure the cursor was still blinking, it continued on and finished up. Note I'm on a RPi4B, but I believe the step around 84% involves removing the DHCP server. I would expect this to kill SSH if you hadn't previously assigned a static IP address to your RPi.

GitMC-8383, 2023-06-12:
Thanks, so it should work but just be a slower install of pimox. I was getting worried because after 2 hours the cursor was still blinking but it wasn't doing anything. I really needed a VM server because no computer except the raspberry pi can't run the VMware application. I couldn't get VMware ESXi working so I went for a alternative and found this. I will now wait a day or 2 and if it finished up great, if not then I will try to make my own type 1 hypervisor since my pc is the only one that supports VMware without it exploding.

GitMC-8383, 2023-06-12:
Oh, and about your theory about the step around 84% involving removing the DHCP server, I think you were right. I played back the footage from my second attempt form before I gave up, and it turns out you were right. The network tab extension I installed on raspbian before I tried to install pimox went haywire before slowly going into a normal pace. I probably should wait longer before giving up. Right now I am running CasaOS.

GitMC-8383, 2023-06-12:
I am installing it again and once it hit 84% it said the following " THE FOLLOWING PACKAGES WILL BE REMOVED: dhcpcd5" so I came to the conclusion that your theory is correct, the step on 84% is removing the DHCP server.

GitMC-8383, 2023-06-13:
Its been a day and nothing has happened. Still the cursor is blinking so the DHCP server is still uninstalling, but I should still just wait for another day or 2 since my raspberry pi is 1 gb ram edition.

GitMC-8383, 2023-06-13:
I will be using a external drive for the ram since I have an extra drive laying around. I will make the entire drive swap space for proxmox and configure it to the the ram cycle.

GitMC-8383, 2023-06-14:
Update: The installation finished, I am now able to access proxmox and it is functional on raspberry pi 3b+ 1gb ram 16gb ssd

GitMC-8383, 2023-06-14:
Nevermind, this problem just occured.

GitMC-8383, 2023-06-14:
I don't know how to access the proxmox environment. It never displayed it and I can't figure out a way to make it tell me.

dk98174003, 2023-06-24:
I think this is a timeout because the hostname is not changed in /etc/hosts. The new hostname can not be pinged and this is making the delay on 84%.

GitMC-8383, 2023-07-25:
ok i will try it out when i get back form vacation
im currently on vacation so i will haveto wait a week before i can connect to my home network and try it out

AubsUK, 2023-07-25:
With 1GB RAM, you're probably going to struggle, and that's probably why it took so long. It does say requirements are a RPi4

At the end of the install, it tells you how to connect to the web interface:

```
after reboot the PVE web interface will be reachable here :
 --->  https://$RPI_IP_ONLY:8006/  <---
```

If you can't remember, you can plug in a screen and keyboard, log in to the Pi locally, with username `root` and the password you set in the install script.

Type in `ip -c a`, then look for the IP address.

RikiMouraPT, 2024-03-14:
Did you find a fix for this issue?

GitMC-8383, 2024-07-04:
> With 1GB RAM, you're probably going to struggle, and that's probably why it took so long. Also, it does say requirement is an RPi4.
> 
> At the end of the install, it tells you how to connect to the web interface:
> 
> ```
> after reboot the PVE web interface will be reachable here :
>  --->  https://$RPI_IP_ONLY:8006/  <---
> run --->  apt upgrade -y  <---
> in a root shell to complete the installation.
> ```
> 
> If you can't remember, you can plug in a screen and keyboard, log in to the Pi locally, with username `root` and the password you set in the install script.
> 
> Type in `ip -c a`, then look for the IP address.

i have been doing it in keyboard and a monitor and i went on my main pc and typed in the address it sent me and it still didnt work if i remember, also i already got it working so thanks but i just got a rpi3 with 2 gigs of ram and that seemed to fix it

closed 2025-02-23 by GitMC-8383