pimox/pimox7 #145: Bookworm?

martin-braun, 2023-06-25:
Is support for bookworm planned? Do we have to wait for Proxmox to officially ship with Bookworm? There seem to be [official instructions](https://pve.proxmox.com/wiki/Install_Proxmox_VE_on_Debian_12_Bookworm) on getting Proxmox to run on Bookworm.

Thanks.

TheBossME, 2023-06-25:
Read my comments, there is a stable proxmox 8.0-3 for arm64

Search for arm64 unofficial proxmox

Good luck.

martin-braun, 2023-06-26:
Very helpful @TheBossME, dankeschön. :)

TheBossME, 2023-06-26:
Hi Martin,

I will create a YT Video with all the needed steps to prepare a stable Proxmox installation while adding

deb https://mirrors.apqa.cn/proxmox/ pvearm8 main

Repository. I'm not using the pvetest repo. 

The video will be in German language and maybe with English comments too. 

If you have suggestions or questions, feel free to contact me.

Regards Beatrice

AubsUK, 2023-07-16:
Debian ISO (what I use):
https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/

This is a good video on installing the latest Ubuntu, the exact same principle can be used for Debian.
https://youtu.be/A4OhdsYULfk

A very brief summary and quick steps to get it working

Download the ISO
New VM
Set ID/Name
OS: Do not use media
BIOS: UEFI
Disk 10GB
CPU 2
CPU Type: Host

Hardware >> Remove DC/DVD
Add >> CD/DVD Drive >> Set to SCSI >> Choose ISO
Add >> Serial Port  >> 0 (for initial setup)
Display >> Graphic card >> Serial terminal 0
Options >> Boot order >> CD to the top
Start
Console >> xterm
Install

Close terminal, shutdown VM
Hardware >> CD >> Do not use media
Double click Display >> Default
Serial Port >> Remove

martin-braun, 2023-07-16:
> If you have suggestions or questions, feel free to contact me.

What is the name of your YouTube channel, I'd like to subscribe. I'm excited to see your video guide. :)

cvocvo, 2023-07-17:
> Read my comments, there is a stable proxmox 8.0-3 for arm64
> 
> Search for arm64 unofficial proxmox
> 
> Good luck.
> 
> And use this https://raspi.debian.net/tested/20230612_raspi_4_bookworm.img.xz
> 
> With this repo
> 
> https://github.com/jiangcuo/Proxmox-Arm64

Is it possible to upgrade to this via an existing pimox7 install? Or do we have to do a clean install?

donniedice, 2023-08-04:
I had to manually install "libssl1.1_1.1.0g-2ubuntu4_arm64.deb" and "libtpms0_0.9.2-3.1_arm64.deb" to get the install to work.
wget
https://ubuntu.pkgs.org/18.04/ubuntu-main-arm64/libssl1.1_1.1.0g-2ubuntu4_arm64.deb.html
https://debian.pkgs.org/12/debian-main-arm64/libtpms0_0.9.2-3.1_arm64.deb.html
dpkg
this stuff.

cvocvo, 2023-08-04:
> I had to manually install "libssl1.1_1.1.0g-2ubuntu4_arm64.deb" and "libtpms0_0.9.2-3.1_arm64.deb" to get the install to work. wget [...]

To clarify did you do something like this? (Just trying to figure out the right sequence of steps to upgrade/migrate)
 - Have an existing pimox7 install
 - Then upgrade debian from 11 to 12 (and install those packages you linked)
 - Then change the Proxmox repos from the Pimox7 ones to the ones listed here: https://github.com/jiangcuo/Proxmox-Arm64/wiki/Install-Proxmox-VE-on-Debian-bullseye#install-proxmox-ve
 - Then updated packages (thus upgrading proxmox to v8)

donniedice, 2023-08-05:
Ok ignore that, I was doing something wrong. I ended up getting things running using 2023-05-03-raspios-bullseye-arm64-lite.img.xz and Virtual Environment 7.4-16.

Davidorado1, 2023-08-25:
I got it working too. I used mainly [this](https://gist.github.com/gtx28/059ac9a0c85e722225c6d5150d659dec) manual to do it.
But I changed the order a bit:
In one of my two raspi/Proxmox Setups I got an error while upgrading from bullseye to bookworm that I apparently wanted to uninstall proxmox. I stopped at this point and first changed the proxmox repository from the pimox7 to the other one:
deb https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm port
After the next dist-upgrade, apt autoremove, reboot and another apt upgrade everything went really well.

martin-braun, 2023-08-25:
@prafiles @Davidorado1 Not trying to imply something personal about you guys, but you both seem to emphasize that only the Chinese mirror is working. It's a mirror though, it shouldn't make a difference. This looks suspicious. I wouldn't want to take a mirror from China, not because of their people, but because of their rulers.

Someone should spend some time to figure out why the Chinese mirror is working for you. I'll rather try my luck with western mirrors and figure out why it doesn't work.

Davidorado1, 2023-08-25:
Well, I understand your concerns, but why do you come across this issue 2 months later? 
@TheBossME posted [this](https://github.com/jiangcuo/Proxmox-Arm64) github repository in his first answer. Did you even take a look at the repo and the wikipage? The guy behind the repository is the one who provides this mirror (as you can read [here](https://github.com/jiangcuo/Proxmox-Arm64/issues/37) ).
Sure, it can be suspicious, but that shouldn´t be new to you.
Have a nice weekend!

martin-braun, 2023-08-25:
@Davidorado1 I understand where you are coming from. Well I wasn't really concerned about that 2 months earlier until I gained some more knowledge about the current situation over there. But you are right, the second link explains the mirror situation, which isn't a mirror by this definition. I didn't look into its details back then as I'm still busy on other stuff.

Thanks for sharing this with me though. Hopefully jiangcuo is legit.

martin-braun, 2023-08-29:
@prafiles Thank you for the feedback, which confirms my conclusion. Although called "mirror", it's a custom repository adjusted by jiangcuo to work with the PI. I investigated his Chinese blog, he seems to be legit, however we know how relationships between companies and governments are quite different in China compared to the Western world, we wouldn't even noticed interference by higher ups, unless you have the time and dedication to audit all the package sources in that "mirror".

I will keep this open for the sake of finding "official" alternatives (as official as it can get).