pimox/pimox7 #146: I try to install vm but i got only errors

Xeno367, 2023-07-02:
I tried to install ubuntu desktop and I got only errors
same for windows
what OS can I install to have a desktop 
What OS can I install to build my own cloud ? with openmediavault/nextcloud/containers

AubsUK, 2023-07-15:
If you need help, you need to give more information, like what errors are you getting? What version OS are you installing?

As you're using Pimox, you'll most likely be on a Raspberry Pi, or at least on an Arm CPU, so you will need to use Arm64 installation ISO.

You can find it here for Ubuntu:
https://cdimage.ubuntu.com/jammy/daily-live/current/

Or here for Debian (what I use):
https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/

This is a good video on installing the latest Ubuntu, the exact same principle can be used for Debian.

https://youtu.be/A4OhdsYULfk

GitMC-8383, 2023-07-25:
i already have an issue open for an issue im experiencing with pimox

kinhsman, 2023-10-31:
> If you need help, you need to give more information, like what errors are you getting? What version OS are you installing?
> 
> As you're using Pimox, you'll most likely be on a Raspberry Pi, or at least on an Arm CPU, so you will need to use Arm64 installation ISO.
> 
> You can find it here for Ubuntu: https://cdimage.ubuntu.com/jammy/daily-live/current/
> 
> Or here for Debian (what I use): https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/
> 
> This is a good video on installing the latest Ubuntu, the exact same principle can be used for Debian.
> 
> https://youtu.be/A4OhdsYULfk
> 
> A very brief summary and quick steps to get it working
> 
> Download the ISO New VM Set ID/Name OS: Do not use media BIOS: UEFI Disk 10GB CPU 2 CPU Type: Host
> 
> Hardware >> Remove DC/DVD Add >> CD/DVD Drive >> Set to SCSI >> Choose ISO Add >> Serial Port >> 0 (for initial setup) Display >> Graphic card >> Serial terminal 0 Options >> Boot order >> CD to the top Start Console >> xterm Install
> 
> Close terminal, shutdown VM Hardware >> CD >> Do not use media Double click Display >> Default Serial Port >> Remove

I followed your steps and got the following error:
`TASK ERROR: KVM virtualisation configured, but not available. Either disable in VM configuration or enable in BIOS.`

Then I disabled KVM virtualization inder the Options, it returned the following error:
```
kvm: The 'host' CPU type can only be used with KVM or HVF
TASK ERROR: start failed: QEMU exited with code 1
```

Mine is NanoPi R6S running ARMv8 CPU
Any insight is greatly appreciated!

TheBossME, 2023-10-31:
Please post your .conf file for the VM to check. Thankx

kinhsman, 2023-10-31:
GNU nano 5.4                    /etc/pve/nodes/nanopi-r6s-pve/qemu-server/106.conf                             
bios: ovmf
boot: order=scsi2;scsi0;net0
cores: 2
cpu: host
efidisk0: local:106/vm-106-disk-2.qcow2,efitype=4m,size=64M
kvm: 1
memory: 2048
meta: creation-qemu=7.0.0,ctime=1698762086
name: ubuntu
net0: virtio=9E:9F:EF:16:F3:D4,bridge=vmbr0
numa: 0
ostype: l26
scsi0: local:106/vm-106-disk-1.qcow2,size=10G
scsi2: synology-hdd:iso/jammy-desktop-arm64.iso,media=cdrom,size=3597740K
scsihw: virtio-scsi-pci
serial0: socket
smbios1: uuid=8576eb53-6866-46f8-bf43-e5d2dcec88da
sockets: 1
vga: serial0