pimox/pimox7 #148: USB 1.1 device problem

bobmorane83, 2023-07-25:
Hi all,

I'am facing a problem trying to mount USB 1.1 devices (but I've no problem with USB 2.0 devices !) on my RPi 4.
I've tried on two differents VM : 
* Ubuntu Jammy Cloud Image
* OpenWRT latest self compiled.

When I plug a FTDI TTL232R USB Serial device, here is what I get :
(I've tested on USB2 and USB3 Pi connector).

```
root@RPi4-01-PVE:~# qm monitor 101
Entering Qemu Monitor for VM 101 - type 'help' for help

qm> info usbhost
  Bus 1, Addr 17, Port 1.3, Speed 12 Mb/s
    Class 00: USB device 0403:6001, TTL232R-3V3
  Bus 1, Addr 11, Port 1.1, Speed 480 Mb/s
    Class ef: USB device 8087:095a, MODEM + 2 CDC-ACM + 3 CDC-NCM + SS

qm> info usb
  Device 0.0, Port 1, Speed 12 Mb/s, Product TTL232R-3V3, ID: usb0
```
So the device is seen by the host and seems to be passed through to the guest.

But in the VM Guest : 

```
ubuntu@Jammy:~$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
and nothing at all about the device in dmesg : 
```
...
[    1.024842] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.027034] ehci-pci: EHCI PCI platform driver
[    1.028971] ACPI: \_SB_.PCI0.GSI0: Enabled at IRQ 35
[    1.040261] ehci-pci 0000:00:01.0: EHCI Host Controller
[    1.067562] ehci-pci 0000:00:01.0: new USB bus registered, assigned bus number 1
[    1.079407] ehci-pci 0000:00:01.0: irq 45, io mem 0x10040000
[    1.095129] ehci-pci 0000:00:01.0: USB 2.0 started, EHCI 1.00
[    1.097366] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
[    1.099993] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.102130] usb usb1: Product: EHCI Host Controller
[    1.103687] usb usb1: Manufacturer: Linux 5.15.0-75-generic ehci_hcd
[    1.105562] usb usb1: SerialNumber: 0000:00:01.0
[    1.119302] hub 1-0:1.0: USB hub found
[    1.120541] hub 1-0:1.0: 6 ports detected
[    1.127246] ehci-orion: EHCI orion driver
[    1.128655] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    1.130557] ohci-pci: OHCI PCI platform driver
[    1.134840] uhci_hcd: USB Universal Host Controller Interface driver
...
```
Any clue ?

Thanks,
BR

TheBossME, 2023-07-25:
Hi Vincent,Which kernel and pve version on pve node ?. I‘m actually using 6.4 and 6.1 kernel and PVE 8.0.3 updated todayI‘m using the stable and tested Debian port (not RaspOS).From node:info usbhost  Bus 1, Addr 3, Port 1.4, Speed 12 Mb/s    Class 00: USB device 0572:0001From VM: lsusbBus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hubBus 003 Device 002: ID 0572:0001 Conexant Systems (Rockwell), Inc. Ezcam II WebcamBus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hubBus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hubBus 001 Device 003: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB KeyboardBus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB TabletBus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hubRegardsBeatrice

bobmorane83, 2023-07-25:
Hi Beatrice,

Here the versions :
* Pimox
PVE Manager Version : pve-manager/7.2-7/f1b67e2b

* Host : 
Debian bulleye 11
```
root@RPi4-01-PVE:~# uname -a
Linux RPi4-01-PVE 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

```
* Guest : 
Ubuntu Jammy 22.04.02 LTS
```
root@Jammy:~# uname -a
Linux Jammy 5.15.0-75-generic #82-Ubuntu SMP Mon Jun 12 06:02:05 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
```

BTW, how did you managed to install PVE 8 on Raspberry Pi  ?

Thanks for your help,
V.

TheBossME, 2023-07-25:
Time to upgrade:Proxmox-Port/help/repo.md at main · jiangcuo/Proxmox-Portgithub.comI prefer fresh install with Debian port and not raspos.Keep in mind, it isn‘t official repo.

bobmorane83, 2023-07-25:
I'll upgrade and let you know.
Thx.