pimox/pimox7 #150: issues starting up CL

sanderweg, 2023-07-29:
hi, 

i'm fairly new to proxmox but i already manage to build a cluster withe 4 Pi's.
now i'm trying to build a CL but it keeps crasing on boot.
in de console line it gives me this error code

lxc-console: 101: ../src/lxc/tools/lxc_console.c: main: 129 101 is not running

TheBossME, 2023-07-30:
I have setup cluster with 2 arm devices with iscsi storage and 1 amd device als quorum.

Let me know your exact setup. 

Are you running Proxmox 7 oder Proxmox 8 ?
Ist it plain Debian used or RaspOS with chroot kernel ?

I also setup a similar config for somebody else and it‘s running with success
On 7.4.3 and 8.0.3 Proxmox Arm architecture.

Feel free to,contact me to make a fresh clean and running setup.

Regards
Beatrice

sanderweg, 2023-07-30:
hi 

i'm running the proxmox 7 version (7.2 to be exact). the OS that are running on the Pi's are the debianbullseye from the rasbarry image aplication. i have not attache a storage device yet so i still run the micro sd cards withe 64 gb. 
i'm planning to do in the future.

TheBossME, 2023-07-30:
Building a Cluster without a shared storage for the nodes isn‘t possible.
hi
i'm running the proxmox 7 version (7.2 to be exact). the OS that are running on the Pi's are the debianbullseye from the rasbarry image aplication. i have not attache a storage device yet so i still run the micro sd cards withe 64 gb.
i'm planning to do in the future.

sanderweg, 2023-07-30:
i can't use the local storage that is on 1 of my pi"s ? i have a synology Nas where i tried to make a shared storrage but that wasn't going as planned

TheBossME, 2023-07-31:
Create an iSCSI LUN in needed size for the cluster on the Synology NAS if there is a 1GB network connection on your LAN. 

Use VLAN‘s if possible in your network for the cluster communication. (Only recommendation).

Install open-iscsi on both nodes (if not installed automatically)
Run scsi discovery on the nodes (keep in mind, that you have min 3 nodes for quorum)

Change the manual startup for iSCSI setting to Automatic on all nodes
use only static ip addresses (no SLAAC) for the iscsi connection.
Do prefer ipv4 static.

To,see the iscsi LUNs on the Proxmox nodes, you can use the iscsi option in web interface, but I do prefer cli only on the nodes.

Good luck 👍🍀

sanderweg, 2023-07-31:
aah thank you. i manage to make a shared storage trough ISCSIs. but i still get the error when i want to boot up my first container 

lxc-console: 100: ../src/lxc/tools/lxc_console.c: main: 129 100 is not running

TheBossME, 2023-07-31:
do you get that rootfs image from here ?Index of /images/debian/bookworm/arm64/default/20230731_05:24images.canonical.comIf yes, keep in mind to not setup dhcp and slaac before „apt update“ and installing ifupdown package from repo into the LXC container.start with static and no ip setup first.

TheBossME, 2023-08-04:
Correct. 

If you like to contact me on discord 

https://discord.gg/rpicloud

sanderweg, 2023-08-04:
thank you, i added you on discord.

sanderweg, 2023-08-04:
thank you, i added you on discord. now i'm trying to use bookworm but every time im making the new sd card it doesn't remember my user credentials. is there a standart user account i know nothing about? XD