pimox/pimox7 #156: Pimox completley lost but VM's still working.....

Mattncb, 2023-08-14:
Can anyone suggest how I can try and diagnose this strange issue? This same issue happens every time I retry to install (5+times now)

Pimox succesfully installed and all working ok with a HAOS VM installed and all working ok
Several reboots all cycle ok and Pimox boots up and HAOS VM starts up automatically ok.

Overnight something happens and all access to PIMOX is lost

At the terminal all commands return bash no such file or directory exist including simple sudo ls or ip -a.
remote SSH no longer working
PIMOX IP Address no longer working

BUT

The HAOS VM is still running and accessible on its IP Address!!

Ive reviewed the log files but nothing is being written to it since the day prior and I assume at the point the fault occured.

TheBossME, 2023-08-14:
Which Image do you use ?

I've tried with Proxmox 8.0.3 and this image

https://github.com/home-assistant/operating-system/releases/download/10.4/haos_generic-aarch64-10.4.img.xz

Mattncb, 2023-08-14:
That's the image, but I maybe didn't explain my issue very well,as HAOS is the only thing still working correctly