pimox/pimox7 #157: I can create CT but i can't start it

berkoca, 2023-08-17:
When i try to start the CT that i create, it gives me these errors:
`sync_wait: 34 An error occurred in another process (expected sequence number 7)`
`__lxc_start: 2074 Failed to spawn container "100"`
`TASK ERROR: startup for container '100' failed`

I have Raspberry Pi 4 - 4GB and using the template `debian-11-standard_11.3-1_amd64.tar.zst`

I have formatted the mSD card and installed the pimox the like 10 times, but im getting these errors every time.

What i'm doing wrong?

berkoca, 2023-08-17:
Rookie mistake. I'm reading amd64 as arm64 🤦‍♂️. 

The URL you gave returns error like "debian 12 is not supported", but when i try with bullseye version (i don't know if i can run bookworm version though), it is successful! 

Big thanks @TheBossME!

I have one more question. Can i run distros like alpine, ubuntu, centos etc. or i have to stick with debian?

berkoca, 2023-08-17:
Okay i figured it out that other arm64 distros work as well.

My problem was try to use amd64 instead of arm64.

closed 2023-08-17 by berkoca

TheBossME, 2023-08-18:
Upgrade pimox to version 8.x and Debian bookworm should work then. 

Using this repo.

https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md