pimox/pimox7 #160: Bug: unable to open file '/etc/network/interfaces.tmp.3239' - No such file or directory

C-Fu, 2023-09-18:
I'm installing from Dietpi bullseye, and every single time I reinstall - even with default config.txt dietpi.txt and Automation_Custom_Script.sh files - creating a CT will always result in network interfaces failure.

```
Formatting '/var/lib/vz/images/100/vm-100-disk-0.raw', fmt=raw size=8589934592 preallocation=off
Creating filesystem with 2097152 4k blocks and 524288 inodes
Filesystem UUID: 1de122a8-cc51-4cae-926c-b2f025beba8c
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632
extracting archive '/var/lib/vz/template/cache/Debian11arm64-std-20230830_05:24.tar.xz'
Total bytes read: 344678400 (329MiB, 18MiB/s)
Detected container architecture: arm64
unable to open file '/etc/network/interfaces.tmp.3239' - No such file or directory
TASK ERROR: unable to create CT 100 - error in setup task PVE::LXC::Setup::post_create_hook
```

I'm using Pi400 btw.

antonionardella, 2023-11-16:
I am having the same issue on a rpi4 with dietpi bookworm

Have you found any solutions?

LufyCZ, 2023-12-30:
This apparently happens because the container uses `systemd-network` instead of `ifupdown2`. Disabling the `systemd-network` service, installing `ifupdown2` and enabling the `networking` service makes Proxmox configuration work.

grigio, 2024-04-23:
@LufyCZ I tried that but I still get 

```
unable to open file '/etc/network/interfaces.tmp.31005' - No such file or directory
```

grigio, 2024-04-24:
> Create dir /etc/network manually after you've installed the LXC container without dhcp or ip address.

Thanks. It seems to work adding manually `/etc/network` (should it be /etc/networks ?) the container starts but it still doesn't have internet or a LAN IP, I tried both DHCP and manual

grigio, 2024-04-24:
> Start dhclient manually and do apt install ifupdown

it doesn't give an ip or errors

```

```

```
root@bbbb:~# dhclient eth0
root@bbbb:~# 
root@bbbb:~# ping 1.1.1.1
ping: connect: Network is unreachable
root@bbbb:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0@if47: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether bc:24:11:cb:cd:4e brd ff:ff:ff:ff:ff:ff link-netnsid 0
    inet6 fe80::be24:11ff:fecb:cd4e/64 scope link 
       valid_lft forever preferred_lft forever
```

TheBossME, 2024-04-24:
Why 'dhclient eth0' ?

grigio, 2024-04-25:
@TheBossME I'd like to get a valid LAN ip for the LXC container

```
root@bbbb:~# dhclient -v
Internet Systems Consortium DHCP Client 4.4.3-P1
Copyright 2004-2022 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/bc:24:11:cb:cd:4e
Sending on   LPF/eth0/bc:24:11:cb:cd:4e
Sending on   Socket/fallback
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 14
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 11
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 11
DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 19
No DHCPOFFERS received.
No working leases in persistent database - sleeping.
```

grigio, 2024-04-25:
@TheBossME I used proxmox x86 in the past and now i'm on proxmox 8 arm64 unofficial.. and i didnt have this issue, however thanks, i'll try other paths

FunDeckHermit, 2025-03-08:
Thanks! This helped me start the Debian Trixie container!

pnward, 2025-06-12:
# Configuring a Portable Pimox Container with WiFi Bridge

This guide explains how to set up a Debian 11 container in Pimox (a Raspberry Pi ARM64 fork of Proxmox) with a WiFi bridge for portability, suitable for use in hotels or other dynamic network environments. It covers configuring a static IP and ensuring network connectivity.

**Note**: Bridging WiFi (wlan0) is unconventional and not always recommended, but this setup works for a portable, multi-purpose device with a desktop GUI.

## Prerequisites

- Pimox installed with `ifupdown2` package.
- WiFi interface (wlan0) configured, either via GUI or manually.
- A bridged network setup (vmbr0) on the host.
- Basic familiarity with Linux networking commands.

## Step 1: Create the Container

1. **Download Debian 11**: Select Debian 11 as the container template in Pimox.
2. **Create Container**:
   - In the Pimox GUI, create a new container.
   - During network setup, choose **static** but leave IP and Gateway fields blank.
   - Confirm the container builds without errors.

## Step 2: Configure the Pimox Host Network

The host uses a bridge (vmbr0) to route traffic through the WiFi interface (wlan0). My setup uses:
- **Pimox bridge IP**: `192.168.88.99/24`
- **Home router IP/DHCP range**: `192.168.1.1`

### Host Network Configuration

Edit `/etc/network/interfaces` on the Pimox host:

```bash
auto lo
iface lo inet loopback

auto eth0
iface eth0 inet dhcp

auto vmbr0
iface vmbr0 inet static
    address 192.168.88.99/24
    bridge-ports none
    bridge-stp off
    bridge-fd 0
    post-up   echo 1 > /proc/sys/net/ipv4/ip_forward
    post-up   iptables -t nat -A POSTROUTING -s '192.168.88.0/24' -o wlan0 -j MASQUERADE
    post-down iptables -t nat -D POSTROUTING -s '192.168.88.0/24' -o wlan0 -j MASQUERADE
    post-up   iptables -t raw -I PREROUTING -i fwbr+ -j CT --zone 1
    post-down iptables -t raw -D PREROUTING -i fwbr+ -j CT --zone 1
```

**Notes**:
- Enable the bridge in the Pimox GUI to ensure reliable startup.
- If using a wired connection, replace `wlan0` with `eth0` in the configuration.
- The `MASQUERADE` rule ensures traffic from the `192.168.88.0/24` network routes correctly through wlan0.

## Step 3: Start and Configure the Container

1. **Start the Container**: Log in via the Pimox console or SSH.
2. **Set a Static IP**: Assign an IP to the container on the `192.168.88.0/24` network. Run the following commands:

```bash
# Assign IP to eth0
ifconfig eth0 192.168.88.200 netmask 255.255.255.0 up

# Set default route
ip route add default via 192.168.88.99 dev eth0

# Verify network settings
ip a
ip r

# Test connectivity
ping 192.168.1.1   # Router
ping 192.168.88.99 # Pimox host
ping 1.1.1.1       # External IP
```

3. **Update the System**: If connectivity is confirmed, update the container:

```bash
apt update && apt -y upgrade
apt install ifupdown2
```

4. **Persist Network Settings**: Edit `/etc/network/interfaces` in the container to make the IP configuration permanent:

```bash
auto lo
iface lo inet loopback
iface lo inet6 loopback

auto eth0
iface eth0 inet static
    address 192.168.88.200/24
    gateway 192.168.88.99
```

## Step 4: Finalize in Pimox GUI

- In the Pimox GUI, assign an unused IP (e.g., `192.168.88.200`) and gateway (`192.168.88.99`) to the container.
- Stop and restart the container to apply changes.

## Step 5: Verify Everything

- Check `/etc/network/interfaces` in the container to confirm settings.
- Test connectivity again with `ping` commands.
- Ensure the container works after a reboot.

## Notes

- This setup assumes a desktop GUI for wlan0 configuration. For headless systems, configure wlan0 in `/etc/network/interfaces` or use a wired connection (eth0).
- The `192.168.88.0/24` subnet is used for the Pimox bridge to avoid conflicts with the router’s DHCP range (`192.168.1.0/24`).
- If connectivity issues arise, double-check iptables rules and bridge settings.

Hope this helps! Let me know if you run into issues or need clarification.