pimox/pimox7 #164: pimox freezing far to often

Yngve0, 2023-10-14:
I am running pimox7.2.7 on  rpi 4 8GB in a 3 node-cluster.

It seems to run stable most of the time, but from time to time one of the nodes with its vm's freeze for approx 30 min. The actual node respons on ping but is not available on "native ssh" or proxmox - webUi. After around 30 minutes it is back in business and "uptime" on the node or its guest OS indicate a reboot.

The average load is around 5GB RAM and under 60% CPU on all nodes.

Any advice is welcome.

albertogrimana, 2023-10-15:
Hi!!
I had same problem and I fix it synchrony datetime of cluster with NTP and permanent with a RTC module clock

TheBossME, 2023-10-15:
chrony installed ?

Yngve0, 2023-10-20:
>>chrony installed ?
Yes

>>I had same problem and I fix it synchrony datetime of cluster with NTP and permanent with a RTC module clock
              Local time: Fri 2023-10-20 06:04:51 BST
           Universal time: Fri 2023-10-20 05:04:51 UTC
                 RTC time: n/a
                Time zone: Europe/London (BST, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

TheBossME, 2023-10-20:
I didn't get issues with my raspi cluster installations.

I use raspi.Debian.net images without issues.
More,then 100 times without having issues, but I removed ifupdown config and timesyncd config files completely with purge. Chrony worked fine then. 

I didn't need the raspberry specific overhead and packages.

Yngve0, 2023-10-20:
> I didn't get issues with my raspi cluster installations.
> 
> I use raspi.Debian.net bookworm images without issues. More,then 100 times without having issues, but I removed ifupdown config and timesyncd config files completely with purge. Chrony worked fine then.
> 
> I didn't need the raspberry specific overhead and packages.

Will pimox7 run on other then bullseye?