pimox/pimox7 #165: Can't get Pimox to work on my Rpi4

peter9030, 2023-10-20:
After running the script I get the following.

[image]

After reboot the webinterface is not found.

TheBossME, 2023-10-20:
failed installation. Do it again with proper image (bullseye)

peter9030, 2023-10-20:
Thanks for your answer!
I used Bookworm as it's the latest release for RPi.
Is there a way to use Bookworm instead of Bullseye?

TheBossME, 2023-10-20:
Yes, of course 

https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md

If you're using official bookworm image, please 

Add [arch=arm64] in the pveport.list for proper installation

peter9030, 2023-10-20:
Thanks a lot! I have it working using Bullseye. I'll have a look soon to see if I can make it work using jiangcuo repo

gitgrimbo, 2023-10-22:
@peter9030, did you find a way to use bookworm? Trying and failing to install on rpi 5 with bookworm.

TheBossME, 2023-10-23:
Go to /etc/apt/sources.list.d/pveport.list and change to 

deb [arch=arm64] https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm port