pimox/pimox7 #171: TASK ERROR: start failed: QEMU exited with code 1

WhiTe-Yak, 2023-11-23:
kvm: -device nec-usb-xhci,id=xhci,bus=pcie.1,addr=0x1b: Bus 'pcie.1' not found
TASK ERROR: start failed: QEMU exited with code 1

kvm: -device ide-cd,bus=ide.0,unit=0,drive=drive-ide0,id=ide0,bootindex=100: Bus 'ide.0' not found
TASK ERROR: start failed: QEMU exited with code 1

Error with all VM

EndroZed, 2023-11-24:
Do not try to load via the templates, they are based on x86. Try to download an operating system from the official sites

TheBossME, 2023-11-24:
https://jenkins.linuxcontainers.org

Use it from here

dstamou, 2024-01-01:
Getting the same error when I try to connect a usb hardware device. 

I have home assistant set up and can start it without problems if I am not connecting a hardware device.

Error only occurs after I add a "USB Device".

What could be the reason? Any idea how I can solve this?

[image]

danhxnk, 2024-01-11:
> Getting the same error when I try to connect a usb hardware device.
> 
> I have home assistant set up and can start it without problems if I am not connecting a hardware device.
> 
> Error only occurs after I add a "USB Device".
> 
> What could be the reason? Any idea how I can solve this?
> 
> ![CleanShot 2024-01-01 in Google Chrome from pimox - Proxmox Virtual Environment@2x](https://private-user-images.githubusercontent.com/7046503/293607561-004d0d4a-be99-4bbe-9d35-d96654fb7189.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMDI1ODAsIm5iZiI6MTcwNTAwMjI4MCwicGF0aCI6Ii83MDQ2NTAzLzI5MzYwNzU2MS0wMDRkMGQ0YS1iZTk5LTRiYmUtOWQzNS1kOTY2NTRmYjcxODkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDExMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMTFUMTk0NDQwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGM3MTNkODM3OTBkZTU3M2YyZmVjNzA1NzgxY2E5NGRlNmIzMDkwNTIzMTJhZjUxMWUwMTIzZGFlNjM5YjQzNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.tmY9KnBfh2jt22trPLnVbiEGraGDlqtvhF5Po4UD-Bc)

I had the same issue, but unchecking "Use USB3" allows the VM to boot. Home Assistant doesn't seem to recognise the new device, but that could be a diferent issue.