pimox/pimox7 #172: unmet dependencies

kyung-yeon, 2023-11-27:
unmet dependencies error when run `./RPiOS64-IA-Install.sh`

```
The following packages have unmet dependencies:
 ceph-fuse : Depends: libfmt7 (>= 7.1.3+ds1) but it is not installable
 libpve-rs-perl : Depends: perlapi-5.32.1 but it is not installable
 libpve-u2f-server-perl : Depends: perlapi-5.32.1 but it is not installable
 librados2-perl : Depends: perlapi-5.32.1 but it is not installable
 lxc-pve : Depends: libgnutlsxx28 but it is not installable
 proxmox-ve : Depends: pve-qemu-kvm but it is not installable
 pve-cluster : Depends: perlapi-5.32.1 but it is not installable
 qemu-server : Depends: pve-qemu-kvm (>= 3.0.1-62) but it is not installable
 spiceterm : Depends: pve-qemu-kvm but it is not installable
E: Unable to correct problems, you have held broken packages.

=========================================================================================
 ! FIXING NETWORK CONFIGURATION.... ERRORS ARE NOMALAY FINE AND RESOLVED AFTER REBOOT !
=========================================================================================

cp: cannot stat '/usr/share/javascript/proxmox-widget-toolkit/proxmoxlib.js': No such file or directory
sed: can't read /usr/share/javascript/proxmox-widget-toolkit/proxmoxlib.js: No such file or directory
```

TheBossME, 2023-11-27:
Please check the prereq's 

Pimox7 is on Debian bullseye. You've tried it with bookworm. This will not work.

kyung-yeon, 2023-11-27:
I installed it using the Raspberry Pi tool.

device: raspeberry pi 4
os : raspberry pi os lite (64-bit)

TheBossME, 2023-11-27:
You didn't select the bullseye image for pimox.

As alternative use bookworm with a different repo.

https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md

Follow the instructions and add [arch=arm64] in the repo file like pveport.list

Good luck 👍

johnhart96, 2023-12-07:
I am having this problem on bullseye

RFDarter, 2024-01-01:
`
The following packages have unmet dependencies:
 ceph-fuse : Depends: libfmt7 (>= 7.1.3+ds1) but it is not installable
 libpve-rs-perl : Depends: perlapi-5.32.1 but it is not installable
 libpve-u2f-server-perl : Depends: perlapi-5.32.1 but it is not installable
 librados2-perl : Depends: perlapi-5.32.1 but it is not installable
 lxc-pve : Depends: libgnutlsxx28 but it is not installable
 proxmox-ve : Depends: pve-qemu-kvm but it is not installable
 pve-cluster : Depends: perlapi-5.32.1 but it is not installable
 qemu-server : Depends: pve-qemu-kvm (>= 3.0.1-62) but it is not installable
 spiceterm : Depends: pve-qemu-kvm but it is not installable
E: Unable to correct problems, you have held broken packages.
`
Same for me.
A fresh image of the 64 bit version of the raspberry pi os lite was flashed on a sd card.
apt update and apt upgrade was exectuted befor using the script.

NeK000, 2024-01-01:
Same issue here. 
root@pimox-1:/home/kube# cat  /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Output of the install 
Package pve-manager is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libpve-guest-common-perl

E: Package 'pve-manager' has no installation candidate
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 proxmox-ve : Depends: pve-manager but it is not installable
              Depends: pve-qemu-kvm but it is not installable
              Depends: qemu-server but it is not installable
              Depends: spiceterm but it is not installable
              Depends: vncterm but it is not installable
              Depends: chrony but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

=========================================================================================
 ! FIXING NETWORK CONFIGURATION.... ERRORS ARE NOMALAY FINE AND RESOLVED AFTER REBOOT !
=========================================================================================

mrfatbat, 2024-11-28:
OK so I got this as well.  I got it working with the manual install but I HAD TO USE this version of [raspios](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-05-03/2023-05-03-raspios-bullseye-arm64-lite.img.xz). **2023-05-03-raspios-bullseye-arm64-lite.img.xz**.

mrfatbat, 2024-11-28:
Other later versions of bullseye didn't work.  230503 did.