pimox/pimox7 #173: Raspberry PI5

Derjeff05, 2023-12-09:
Hey i got a Problem caused by the new Kernel of Raspi 5.
Will there be an update for Raspi5? Cause its annoing thst it just fail by the new Kernel.

Thanks for ur Time and Work on here.

TheBossME, 2023-12-10:
Is,there a bullseye image for Raspi 5 ?.

If not, don't use pimox 7

Derjeff05, 2023-12-10:
U did not understand my question. No u can only use Debian 12 and Ubuntu 23.1 with raspi5 my question was. If the programmer of this library will program a new library for Debian 12 or Ubuntu 23.1.

Derjeff05, 2023-12-10:
And also maybe Proxmox 8

TheBossME, 2023-12-10:
Yes. I described it a lot

Derjeff05, 2023-12-10:
Bro what? What did u described? Can u pls write more than one sentence? lol haha I don't mean it mean but until now the only sense in ur sentence was like. Pimox7 ist not avaible on raspi5. Like yes I got it dude. But maybe there will be one day an library on GitHub with pimox8 and then this will be abaible for raspi5. And I just asked the owners of this library.

SabrinaRDC, 2023-12-11:
@Derjeff05 did you checkthe issue page where basically the exact same question was asked not long ago? Because that thread probably answered your question about how to set up proxmox on a rpi5.

hellresistor, 2023-12-23:
have tried using this image ?
https://raspi.debian.net/tested/20231109_raspi_4_bookworm.img.xz

epsimatic, 2023-12-28:
FYI: [jiangcuo/Proxmox-Port](https://github.com/jiangcuo/Proxmox-Port) seems to support pve8 base on debian12(arm64)

zieglerjm, 2024-01-07:
Has anyone gotten it installed on Bookworm server?  So far I gant get past errors:

- apt install proxmox-ve

The following packages have unmet dependencies:
 ceph-fuse : Depends: libfmt7 (>= 7.1.3+ds1) but it is not installable
 libpve-rs-perl : Depends: perlapi-5.32.1 but it is not installable
 libpve-u2f-server-perl : Depends: perlapi-5.32.1 but it is not installable
 librados2-perl : Depends: perlapi-5.32.1 but it is not installable
 lxc-pve : Depends: libgnutlsxx28 but it is not installable
 proxmox-ve : Depends: pve-qemu-kvm but it is not installable
 pve-cluster : Depends: perlapi-5.32.1 but it is not installable
 qemu-server : Depends: pve-qemu-kvm (>= 3.0.1-62) but it is not installable
 spiceterm : Depends: pve-qemu-kvm but it is not installable
E: Unable to correct problems, you have held broken packages.

[image]

TheBossME, 2024-01-07:
Using bullseye done.

zieglerjm, 2024-01-07:
Thanks, [Install-Proxmox-VE-on-Debian-bookworm-pve8-(ARM64)](https://github.com/jiangcuo/Proxmox-Port/wiki/Install-Proxmox-VE-on-Debian-bookworm) worked for me but it never asked for a ROOT password?  Does this script use a default value?