pimox/pimox7 #181: Cannot start container

GitHubElory, 2024-04-06:
I have installed pimox on Raspberry Pi OS (Legacy, 64-bit), but I got this error on while starting an LXC container
[image]

TheBossME, 2024-04-06:
why are all the instruments that search for intelligent life being directed away from the earth ?

if you're using different processor architecture, what will be the result ?

You'ge got it !

GitHubElory, 2024-04-06:
[image]

 I am using arm64...

GitHubElory, 2024-04-06:
> why are all the instruments that search for intelligent life being directed away from the earth ?
> 
> if you're using different processor architecture, what will be the result ?
> 
> You've got it !

Can you tell me where I can find arm64 container templates?

TheBossME, 2024-04-06:
[image]

GitHubElory, 2024-04-07:
My proxmox version is 7.2, and I have no idea how to update it.

TheBossME, 2024-04-07:
https://github.com/jiangcuo/Proxmox-Port/wiki/Install-or-upgrade-from-repo

GitHubElory, 2024-04-07:
Thank you, I updated proxmox to 8.1.7 however...
I don't know how to properly create a linux bridge could you help me with that also?

When I'm creating a bridge it disconnects me with web gui and I can't do anything.
[image]
[image]
and after I click apply edits I cannot reconnect.

TheBossME, 2024-04-07:
[image]

[image]

Do,it via web interface

modify your Hosts entry properly if not done.

Then create a bridge without an ip address first,and apply.

Then remove the ip address from eth0 and put it to vmbr0 then.

Good luck 🍀

GitHubElory, 2024-04-07:
It worked, I mean kind of...

[image]

when I'm using DHCP on bridge then it just doesn't give me an address
[image]
[image]

However when I'm giving it an static address it just pinging forever without response
[image]
[image]
[image]

The strange thing is that the address show up in router 
[image]

GitHubElory, 2024-05-10:
I had to uninstall a docker ¯\_(ツ)_/¯ to fix it

closed 2024-05-10 by GitHubElory