pimox/pimox7 #183: broken packages

Monnorthegodlydev, 2024-10-15:
`Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Get:4 https://raw.githubusercontent.com/pimox/pimox7/master dev/ InRelease [2,186 B]
Hit:5 http://archive.raspberrypi.com/debian bookworm InRelease                                
Get:6 https://raw.githubusercontent.com/pimox/pimox7/master dev/ Packages [101 kB]
Fetched 103 kB in 2s (65.5 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: https://raw.githubusercontent.com/pimox/pimox7/master/dev/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  chromium-browser chromium-browser-l10n chromium-codecs-ffmpeg-extra
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'dhcpcd5' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  chromium-browser chromium-browser-l10n chromium-codecs-ffmpeg-extra
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  chromium-browser chromium-browser-l10n chromium-codecs-ffmpeg-extra
0 upgraded, 0 newly installed, 3 to remove and 0 not upgraded.
After this operation, 4,167 kB disk space will be freed.
(Reading database ... 148714 files and directories currently installed.)
Removing chromium-browser (1:129.0.6668.100-1~deb12u1+rpt1) ...
Removing chromium-browser-l10n (1:129.0.6668.100-1~deb12u1+rpt1) ...
Removing chromium-codecs-ffmpeg-extra (126.0.6478.164-rpt1) ...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ceph-fuse : Depends: libfmt7 (>= 7.1.3+ds1) but it is not installable
 libpve-rs-perl : Depends: perlapi-5.32.1 but it is not installable
 libpve-u2f-server-perl : Depends: perlapi-5.32.1 but it is not installable
 librados2-perl : Depends: perlapi-5.32.1 but it is not installable
 lxc-pve : Depends: libgnutlsxx28 but it is not installable
 pve-cluster : Depends: perlapi-5.32.1 but it is not installable
 pve-qemu-kvm : Depends: liburing1 (>= 0.7) but it is not installable
E: Unable to correct problems, you have held broken packages.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ceph-fuse : Depends: libfmt7 (>= 7.1.3+ds1) but it is not installable
 libpve-rs-perl : Depends: perlapi-5.32.1 but it is not installable
 libpve-u2f-server-perl : Depends: perlapi-5.32.1 but it is not installable
 librados2-perl : Depends: perlapi-5.32.1 but it is not installable
 lxc-pve : Depends: libgnutlsxx28 but it is not installable
 proxmox-ve : Depends: pve-qemu-kvm but it is not installable
 pve-cluster : Depends: perlapi-5.32.1 but it is not installable
 qemu-server : Depends: pve-qemu-kvm (>= 3.0.1-62) but it is not installable
 spiceterm : Depends: pve-qemu-kvm but it is not installable
E: Unable to correct problems, you have held broken packages.
`
I have an RPi-400
I have tried a lot of solutions from other issues but nothing has worked
I have also tried doing the latest ARM release (like the guide says) and with RPiOS 64-bit Lite
Thanks in advance.

TheBossME, 2024-10-15:
Pimox 7 is for bullseye not bookworm. Sorry to say, the issue is YOU !

Monnorthegodlydev, 2024-10-15:
oh, ok. Thanks

closed 2024-10-15 by Monnorthegodlydev