pimox/pimox7 #185: ext journal_check_start:84: comm poxcfs: Detected aborted journal

aatsma, 2025-01-01:
Hello everyone,

Since a couple of days I receive the following error message. It seems like a problem with The Proxmox Cluster file system (“pmxcfs”) and de NVME SSD.

[16821.351985] EX14-fs error (device nvme0n1p2): ext journal_check_start:84: comm poxcfs: Detected aborted journal
[16821.351987] EX14-fs error (device nvme0n1p2) in ext4_reserve_inode_write:5790: Journal has aborted
[16821.371313] EX14-fs (nvme0n1p2): Remounting filesystem read-only

After a hard restart, the system works properly for some time.

Hardware Config:
- Raspberry Pi 5 8GB
- GeeekPi N04 M.2 M-Key NVMe SSD Shield for Raspberry Pi 5, M.2 2280 PCIe to NVMe SSD Shield PiP PCIe Peripheral Board Top for Raspberry Pi 5 4GB/8GB
- WD BLACK SN770 NVMe SSD 1 TB

Software Config:
- PiMox Proxmox Virtual Environment 8.3.1+port1
- Installed 1 VM with HOAS / Home Assistant.

Hopefully someone could help me of guide me in the right direction.

TheBossME, 2025-01-02:
smartctl -a /dev/nvme0n1 ?

aatsma, 2025-01-02:
Hi @TheBossME 

Appreciate your help here. Below the log.

From my perspective I dont see anything strange here. 

=== START OF INFORMATION SECTION ===
Model Number:				WD BLACK SN770 1TB
Serial Number:				24453G403Z68
Firmware Version:			731100WD
PCI Vendor/Subsystem ID:		Ox15b7
IEEE OUI Identifier:			0x001b44
Total NUM Capacity:			1,000,204,886,016 [1,00 TB]
Unallocated NUM Capacity:		0
Controller ID:				0
NVMe Version:				1.4
Number of Namespaces:			1
Namespace 1 Size/Capacity:		1,000,204,886,016 [1.00 TB]
Namespace 1 Formatted LBA Size:		512
Namespace 1 IEEE EUI-64:		001b44 4a41d263e0
Local Time is:				Thu Jan 2 14:57:54 2025 CET
Firmware Updates (0x14):		2 Slots, no Reset required
Optional Admin Commands (0x0017):	Security Format Frmw_DL Self_Test
Optional NUM Commands (0x00df):		Comp Wr_Unc DS_Mngmt Wr Zero Sav/Sel_Feat Timestmp Verify
Log Page Attributes (0x7e):		Cmd_Eff_Lg Ext_Get_Lg Telmtry_Lg Pers_Ev_Lg *Other*
Maximum Data Transfer Size:		256 Pages
Warning Comp. Temp. Threshold:		84 Celsius
Critical Comp. Temp. Threshold:		88 Celsius
Namespace 1 Features (0x02):		NA_Fields

Supported Power States
St Op		Max		Active		Idle 	RL RT WK WT	Ent_Lat	Ex_Lat
0 +		5.00W		5.00W		-	0  0  0  0	0	0
1 +		3.30W		3.00W		-	0  0  0  0	0	0
2 +		2.20W		2.00W		-	0  0  0  0	0	0
3 +		0.0150W		-		-	3  3  3  3	1500	2500
4 +		0.0050W		-		-	4  4  4  4	10000	6000
5 +		0.0033W		-		-	5  5  5  5	176000	25000

Supported LBA Sizes (NSID 0x1)
Id Fmt	Data	Metadt	Rel_Perf
0 +	512	0	2
1 -	4096	0	1

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART Health Information (NVMe Log 0x02)
Critical Warning:			x00
Temperature:				2 Celsius
Available Spare:			100%
Available Spare Threshold:		10%
Percentage Used:			0%
Data Units Read:			39,932 [20.4 GB]
Data Units Written:			100,714 [51.5 GB]
Host Read Commands:			548,500
Host Write Commands:			1,780,934
Controller Busy Time:			9
Power Cycles:				30
Power On Hours:				343
Unsafe Shutdowns:			8
Media and Data Integrity Errors:	0
Error Information Log Entries:		0
Warning omp. Temperature Time:		0
Critical Comp. Temperature Time:	0
Temperature Sensor 1:			43 Celsius
Temperature Sensor 2:			32 Celsius

Error Information (NVMe Log 0x01, 16 of 256 entries)
No Errors Logged 

Thank you in advance!

TheBossME, 2025-01-02:
Try different kernel please

add with nano /etc/apt/sources.list

deb https://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware

try 6.11 backport kernel and do the needed modifications for using backport kernels